NubeIO/driver-bacnet #1: get update name of BO

NubeDev, 2022-05-17:
same as we can update the  `PROP_PRESENT_VALUE`
https://github.com/bacnet-stack/bacnet-stack/blob/ba0cbc1fb8768ff7b915904d836bd29ba32ed64b/src/bacnet/basic/object/bo.c#L365
we want to be able to update this name aswell `PROP_OBJECT_NAME` 
https://github.com/bacnet-stack/bacnet-stack/blob/ba0cbc1fb8768ff7b915904d836bd29ba32ed64b/src/bacnet/basic/object/bo.c#L434

see example of being able to write a string to update the bacnet sever name

read the server name
```
./bacrp 2508 8 2508 77
```
read the server name
```
./bacwp 2508 8 2508 77 0 -1 7 test2
```

shomaglasang, 2022-05-18:
Taking on this @NubeDev . Thanks

shomaglasang, 2022-05-19:
@NubeDev  do I need some permissions to update/resolve the ticket?

shomaglasang, 2022-05-19:
@NubeDev here's the updated bo.c file. Just unzip and drop on src/bacnet/basic/object/bo.c.
[bo.zip](https://github.com/NubeIO/bacnet-server-c/files/8732062/bo.zip)

shomaglasang, 2022-05-29:
Will mark this as completed @NubeDev .
Ready for the next one.

closed 2022-05-29 by shomaglasang