NubeIO/driver-bacnet #3: Supported Objects

NubeDev, 2022-05-29:
init of app objects types and number
the default amount of points will be 0
objects required

- binary input
- binary output
- binary value
- analogue input
- analogue output
- analogue value

so user will put in how many object to create on startup
`BO=25 BI=50`
and so on
The default for each type will be 0
so the command below would only create 50 BIs
```
BI=50 ./bin/bacserv 1234
```

shomaglasang, 2022-06-02:
@NubeDev pushed changes for BI and BO number of object instances on startup. Supported get/set on object property name and present value.
```
BI=10 BO=10 ./bin/bacserv
```

shomaglasang, 2022-06-05:
@NubeDev list objects now supported. Get/set of object name and present value.
```
BI=10 BO=10 BV=10 AI=10 AO=10 AV=10 ./bin/bacserv
```

shomaglasang, 2022-06-07:
@NubeDev Please re-open mate if there are issues.