NubeIO/driver-bacnet #4: Remove/Clen up

NubeDev, 2022-06-03:
to get all the object run this
`./bacepics 1234`

### this is what we want to end up with

- binary input
- binary output
- binary value
- analogue input
- analogue output
- analogue value

### to remove 

- command
- file
- lighting-output
- trend-log
- characterstring-value
- life-safety-point
- octetstring-value
- positive-integer-value
- accumulator

```
    object-list: { 
        (device, 1234), (network-port, 1), (analog-input, 0), 
        (analog-input, 1), (analog-input, 2), (analog-input, 3), 
        (analog-output, 0), (analog-output, 1), (analog-output, 2), 
        (analog-output, 3), (analog-value, 0), (analog-value, 1), 
        (analog-value, 2), (analog-value, 3), (binary-input, 0), 
        (binary-input, 1), (binary-input, 2), (binary-input, 3), 
        (binary-input, 4), (binary-input, 5), (binary-input, 6), 
        (binary-input, 7), (binary-input, 8), (binary-input, 9), 
        (binary-output, 0), (binary-value, 0), (binary-value, 1), 
        (binary-value, 2), (binary-value, 3), (binary-value, 4), 
        (binary-value, 5), (binary-value, 6), (binary-value, 7), 
        (binary-value, 8), (binary-value, 9), (characterstring-value, 0), 
        (command, 0), (command, 1), (command, 2), 
        (command, 3), (integer-value, 1), (notification-class, 0), 
        (notification-class, 1), (life-safety-point, 0), (life-safety-point, 1), 
        (life-safety-point, 2), (life-safety-point, 3), (life-safety-point, 4), 
        (life-safety-point, 5), (life-safety-point, 6), (load-control, 0), 
        (load-control, 1), (load-control, 2), (load-control, 3), 
        (multi-state-input, 0), (multi-state-input, 1), (multi-state-input, 2), 
        (multi-state-input, 3), (multi-state-output, 0), (multi-state-output, 1), 
        (multi-state-output, 2), (multi-state-output, 3), (multi-state-value, 0), 
        (multi-state-value, 1), (multi-state-value, 2), (multi-state-value, 3), 
        (trend-log, 0), (trend-log, 1), (trend-log, 2), 
        (trend-log, 3), (trend-log, 4), (trend-log, 5), 
        (trend-log, 6), (trend-log, 7), (lighting-output, 1), 
        (lighting-output, 2), (lighting-output, 3), (lighting-output, 4), 
        (lighting-output, 5), (lighting-output, 6), (lighting-output, 7), 
        (lighting-output, 8), (channel, 1), (file, 0), 
        (file, 1), (file, 2), (octetstring-value, 0), 
        (octetstring-value, 1), (octetstring-value, 2), (octetstring-value, 3), 
        (positive-integer-value, 0), (positive-integer-value, 1), (positive-integer-value, 2), 
        (positive-integer-value, 3), (schedule, 0), (schedule, 1), 
        (schedule, 2), (schedule, 3), (accumulator, 0), 
        (accumulator, 1), (accumulator, 2), (accumulator, 3), 
        (accumulator, 4), (accumulator, 5), (accumulator, 6), 
        (accumulator, 7), (accumulator, 8), (accumulator, 9), 
        (accumulator, 10), (accumulator, 11), (accumulator, 12), 
        (accumulator, 13), (accumulator, 14), (accumulator, 15), 
        (accumulator, 16), (accumulator, 17), (accumulator, 18), 
        (accumulator, 19), (accumulator, 20), (accumulator, 21), 
        (accumulator, 22), (accumulator, 23), (accumulator, 24), 
        (accumulator, 25), (accumulator, 26), (accumulator, 27), 
        (accumulator, 28), (accumulator, 29), (accumulator, 30), 
        (accumulator, 31), (accumulator, 32), (accumulator, 33), 
        (accumulator, 34), (accumulator, 35), (accumulator, 36), 
        (accumulator, 37), (accumulator, 38), (accumulator, 39), 
        (accumulator, 40), (accumulator, 41), (accumulator, 42), 
        (accumulator, 43), (accumulator, 44), (accumulator, 45), 
        (accumulator, 46), (accumulator, 47), (accumulator, 48), 
        (accumulator, 49), (accumulator, 50), (accumulator, 51), 
        (accumulator, 52), (accumulator, 53), (accumulator, 54), 
        (accumulator, 55), (accumulator, 56), (accumulator, 57), 
        (accumulator, 58), (accumulator, 59), (accumulator, 60), 
        (accumulator, 61), (accumulator, 62), (accumulator, 63) } 

```

shomaglasang, 2022-06-06:
@NubeDev hey mate, I removed the list of items listed above.
this is the result from the query. can you check if the list is good or if there are some that needs to be removed? thanks
```
    object-list: { 
        (device, 1234), (network-port, 1), (analog-input, 0), 
        (analog-input, 1), (analog-input, 2), (analog-input, 3), 
        (analog-input, 4), (analog-input, 5), (analog-input, 6), 
        (analog-input, 7), (analog-input, 8), (analog-input, 9), 
        (analog-output, 0), (analog-output, 1), (analog-output, 2), 
        (analog-output, 3), (analog-output, 4), (analog-output, 5), 
        (analog-output, 6), (analog-output, 7), (analog-output, 8), 
        (analog-output, 9), (analog-value, 0), (analog-value, 1), 
        (analog-value, 2), (analog-value, 3), (analog-value, 4), 
        (analog-value, 5), (analog-value, 6), (analog-value, 7), 
        (analog-value, 8), (analog-value, 9), (binary-input, 0), 
        (binary-input, 1), (binary-input, 2), (binary-input, 3), 
        (binary-input, 4), (binary-input, 5), (binary-input, 6), 
        (binary-input, 7), (binary-input, 8), (binary-input, 9), 
        (binary-output, 0), (binary-output, 1), (binary-output, 2), 
        (binary-output, 3), (binary-output, 4), (binary-output, 5), 
        (binary-output, 6), (binary-output, 7), (binary-output, 8), 
        (binary-output, 9), (binary-value, 0), (binary-value, 1), 
        (binary-value, 2), (binary-value, 3), (binary-value, 4), 
        (binary-value, 5), (binary-value, 6), (binary-value, 7), 
        (binary-value, 8), (binary-value, 9), (integer-value, 1), 
        (notification-class, 0), (notification-class, 1), (load-control, 0), 
        (load-control, 1), (load-control, 2), (load-control, 3), 
        (multi-state-input, 0), (multi-state-input, 1), (multi-state-input, 2), 
        (multi-state-input, 3), (multi-state-output, 0), (multi-state-output, 1), 
        (multi-state-output, 2), (multi-state-output, 3), (multi-state-value, 0), 
        (multi-state-value, 1), (multi-state-value, 2), (multi-state-value, 3), 
        (channel, 1), (schedule, 0), (schedule, 1), 
        (schedule, 2), (schedule, 3) }
```

shomaglasang, 2022-06-07:
@NubeDev mate, pushed the changes to the repo.

shomaglasang, 2022-06-08:
@NubeDev 
List updated.
```
    object-list: { 
        (device, 1234), (network-port, 1), (analog-input, 0), 
        (analog-input, 1), (analog-input, 2), (analog-input, 3), 
        (analog-input, 4), (analog-input, 5), (analog-input, 6), 
        (analog-input, 7), (analog-input, 8), (analog-input, 9), 
        (analog-output, 0), (analog-output, 1), (analog-output, 2), 
        (analog-output, 3), (analog-output, 4), (analog-output, 5), 
        (analog-output, 6), (analog-output, 7), (analog-output, 8), 
        (analog-output, 9), (analog-value, 0), (analog-value, 1), 
        (analog-value, 2), (analog-value, 3), (analog-value, 4), 
        (analog-value, 5), (analog-value, 6), (analog-value, 7), 
        (analog-value, 8), (analog-value, 9), (binary-input, 0), 
        (binary-input, 1), (binary-input, 2), (binary-input, 3), 
        (binary-input, 4), (binary-input, 5), (binary-input, 6), 
        (binary-input, 7), (binary-input, 8), (binary-input, 9), 
        (binary-output, 0), (binary-output, 1), (binary-output, 2), 
        (binary-output, 3), (binary-output, 4), (binary-output, 5), 
        (binary-output, 6), (binary-output, 7), (binary-output, 8), 
        (binary-output, 9), (binary-value, 0), (binary-value, 1), 
        (binary-value, 2), (binary-value, 3), (binary-value, 4), 
        (binary-value, 5), (binary-value, 6), (binary-value, 7), 
        (binary-value, 8), (binary-value, 9) }
```