NubeIO/driver-bacnet #10: Make PROP_RELINQUISH_DEFAULT writeable for AV, AO, BV, BO

NubeDev, 2022-06-08:
https://github.com/NubeIO/bacnet-server-c/blob/e3f3314c55b75cc596027c405914160a4a699f7a/bacnet-stack/src/bacnet/basic/object/bo.c#L505

shomaglasang, 2022-06-12:
@NubeDev hey mate, already implemented this for AO, BV and BO.
I don't think this applies to AV. Please check and let me know.

shomaglasang, 2022-06-13:
Test for BV.

Get present value of object instance 9.
```
./bin/bacrp 1234 5 9 85 --mac 10.104.0.11:47808
inactive
```

Get relinquish default value of object instance 9.
```
./bin/bacrp 1234 5 9 104 --mac 10.104.0.11:47808
inactive
```

Update relinquish default value of object instance 9 to 1 (active).
```
./bin/bacwp 1234 5 9 104 0 -1 9 1 D 10.104.0.11:47808

WriteProperty Acknowledged!
```

Get relinquish default value of object instance 9.
```
./bin/bacrp 1234 5 9 104 --mac 10.104.0.11:47808
active
```

Get present value of object instance 9.
```
./bin/bacrp 1234 5 9 85 --mac 10.104.0.11:47808
active
```

shomaglasang, 2022-06-14:
Test for AO.

Get present value of object instance 9.
```
./bin/bacrp 1234 1 9 85 --mac 10.104.0.11:47808
0.000000
```
Get relinquish default value of object instance 9.
```
./bin/bacrp 1234 1 9 104 --mac 10.104.0.11:47808
0.000000
```
Update relinquish default value of object instance 9 to 11.0
```
/bin/bacwp 1234 1 9 104 0 -1 4 11 D 10.104.0.11:47808
WriteProperty Acknowledged!
```
Get relinquish default value of object instance 9.
```
./bin/bacrp 1234 1 9 104 --mac 10.104.0.11:47808
11.000000
```
Get present value of object instance 9.
```
./bin/bacrp 1234 1 9 85 --mac 10.104.0.11:47808
11.000000
```
Update priority array of object instance 9 at cell 1 to 22.
```
./bin/bacwp 1234 1 9 87 0 1 4 22 D 10.104.0.11:47808
WriteProperty Acknowledged!
```
Get priority array of object instance 9.
```
./bin/bacrp 1234 1 9 87 --mac 10.104.0.11:47808
{22.000000,Null,Null,Null,Null,Null,Null,Null,Null,Null,Null,Null,Null,Null,Null,Null}
```
Get present value of object instance 9.
```
./bin/bacrp 1234 1 9 85  --mac 10.104.0.11:47808
22.000000
```

shomaglasang, 2022-08-31:
@NubeDev mate I will update AV to support PROP_RELINQUISH_DEFAULT.

shomaglasang, 2022-09-04:
Test for AV.

Get present value of object instance 1.
```
./bin/bacrp 1234 2 1 85 --mac 10.104.0.11:47808
0.000000
```
Get relinquish default value of object instance 1.
```
./bin/bacrp 1234 2 1 104 --mac 10.104.0.11:47808
0.000000
```
Update relinquish default value of object instance 1 to 110.22
```
./bin/bacwp 1234 2 1 104 0 -1 4 110.22 D 10.104.0.11:47808
WriteProperty Acknowledged!
```
Get relinquish default value of object instance 1.
```
./bin/bacrp 1234 2 1 85 --mac 10.104.0.11:47808
110.220000
```

shomaglasang, 2022-09-04:
Read/write for AV PROP_RELINQUISH_DEFAULT is already support starting with Release 0.0.17.
Closing this.

closed 2022-09-04 by shomaglasang