NubeIO/driver-bacnet #6: cant write AO above 100 or below 0

NubeDev, 2022-06-08:
https://github.com/NubeIO/bacnet-server-c/blob/10a79451d0084a76e9c5aefedd6758505bbe0e9d/bacnet-stack/src/bacnet/basic/object/ao.c#L223

its this locic i think

shomaglasang, 2022-06-08:
I am not very familiar with the bacnet specs. Here the index and priority are doing the same thing. Currently, BACNET_MAX_PRIORITY is a constant value 16. Should we make this a start-up value? Like
```
AO_MAX_PRIO=100 ./bin/bacserv
```

shomaglasang, 2022-06-12:
@NubeDev , the data type of the storage that holds the value is unsigned char, which means it can only hold a value from 0 to 255.
https://github.com/NubeIO/bacnet-server-c/blob/master/bacnet-stack/src/bacnet/basic/object/ao.c#L63
```
/* Here is our Priority Array.  They are supposed to be Real, but */
/* we don't have that kind of memory, so we will use a single byte */
/* and load a Real for returning the value when asked. */
static uint8_t **Analog_Output_Level = NULL;
```
It is definitely by design. Let me know if you want to change the range of allowed values and what would be the new range? Thanks

shomaglasang, 2022-09-04:
Support writing AO above 100 (e.g. no fixed limit)
```
./bin/bacrp 1234 1 0 85 --mac 10.104.0.11:47808
0.000000
```
```
./bin/bacwp 1234 1 0 85 0 -1 4 1000.20 D 10.104.0.11:47808
WriteProperty Acknowledged!
```
```
./bin/bacrp 1234 1 0 85 --mac 10.104.0.11:47808
1000.200000
```

shomaglasang, 2022-09-04:
Support writing AO below 0.
```
./bin/bacwp 1234 1 0 85 0 -1 4 -21000.30 D 10.104.0.11:47808
WriteProperty Acknowledged!
```
```
./bin/bacrp 1234 1 0 85 --mac 10.104.0.11:47808
-21000.300000
```

shomaglasang, 2022-09-04:
Writing PROP_PRIORITY_ARRAY
```
./bin/bacwp 1234 1 0 87 0 1 4 1000.50 D 10.104.0.11:47808
WriteProperty Acknowledged!
```
```
./bin/bacrp 1234 1 0 87 1  --mac 10.104.0.11:47808
1000.500000
```
```
./bin/bacrp 1234 1 0 85 --mac 10.104.0.11:47808
1000.500000
```

shomaglasang, 2022-09-04:
Fixed writing AO above 100 or below 0.

closed 2022-09-04 by shomaglasang