NubeIO/driver-bacnet #11: setup github action build

NubeDev, 2022-06-17:
ask @RaiBnod if any question

shomaglasang, 2022-06-17:
@NubeDev mate, I created the initial action to build bacnet on push.
I will include test routines as part of the action. Let me know if you want something specific. Thanks

NubeDev, 2022-06-17:
it needs to work for amd64 and armv7

see example
https://github.com/NubeIO/rubix-service/blob/master/.github/workflows/docker-build-release.yml

see relase naming
https://github.com/NubeIO/rubix-service/releases/tag/v1.19.3

```
repo-name | version | build id | target
rubix-service-1.19.3-5824a2bd.amd64.zip
```

shomaglasang, 2022-06-18:
ah yah. will check on that.

shomaglasang, 2022-06-18:
@NubeDev in short, you want a docker image created on push right?

NubeDev, 2022-06-18:
no we dont wanna relase a docker build. We use docker to build the arm version

we want amd64 and arm7

shomaglasang, 2022-06-18:
okay, copy that

shomaglasang, 2022-07-07:
@NubeDev hey mate, the github action is now in place.

amd64
```
$ unzip bacnet-server-c-0.0.1-dd5338df.amd64.zip 
$ file bacserv 
bacserv: ELF 64-bit LSB pie executable, x86-64, version 1 (SYSV), dynamically linked, interpreter /lib64/ld-linux-x86-64.so.2, for GNU/Linux 3.2.0, BuildID[sha1]=9a7745040a21894b5ced859123190d80f0b19516, not stripped
```

arm7
```
$ unzip bacnet-server-c-0.0.1-dd5338df.armv7.zip
$ file bacserv 
bacserv: ELF 32-bit LSB pie executable, ARM, EABI5 version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux-armhf.so.3, for GNU/Linux 3.2.0, BuildID[sha1]=9604893ae38c2d442d77f6d4c654d9812cfd2148, not stripped
```

closed 2022-07-26 by NubeDev