NubeIO/driver-bacnet #15: Write vaue to BI and AI not working

NubeDev, 2022-07-31:


shomaglasang, 2022-08-02:
@NubeDev mate the current implementation is you can only write to PROP_PRESENT_VALUE of BI and AI if the value of PROP_OUT_OF_SERVICE is true (or 1). Like this sequence of commands:
```
$./bin/bacrp 1234 0 0 81 
FALSE

$./bin/bacwp 1234 0 0 81 0 -1 1 1

$./bin/bacrp 1234 0 0 81
TRUE

$./bin/bacwp 1234 0 0 85 0 -1 4 101.22

$./bin/bacrp 1234 0 0 85
101.220000
```

I am not sure if this is how it should be implemented. But the existing code behaves this way. Please let me know if we will change this.

shomaglasang, 2022-08-03:
@NubeDev I removed the checking for out_of_service when updating AI present_value.
```
$./bin/bacrp 1234 0 1 81
FALSE

$./bin/bacwp 1234 0 1 85 0 -1 4 101.50
WriteProperty Acknowledged!

$./bin/bacrp 1234 0 1 85
101.500000
```

closed 2022-08-03 by shomaglasang