NubeIO/driver-bacnet #16: please update these 3 feilds

NubeDev, 2022-08-04:
```
Product Name: "Nube iO rubix"
Product Model Number: "rubix"
Product Description: "nube bacnet server"
```

aidan@pop-os-ryan:~/code/helpers/bacnet-stack-bacnet-stack-1.0.0/bin$ ./bacepics 202
PICS 0
BACnet Protocol Implementation Conformance Statement

--
--
-- Generated by BACnet Protocol Stack library EPICS tool
-- BACnet/IP Interface for BACnet-stack Devices
-- http://sourceforge.net/projects/bacnet/ 
-- 
--

Vendor Name: "Nube iO Operations Pty Ltd"
Product Name: "Nube iO rubix"
Product Model Number: "rubix"
Product Description: "nube bacnet server"

shomaglasang, 2022-08-04:
@NubeDev updated mate.
```
$./bin/bacepics 1234
PICS 0
BACnet Protocol Implementation Conformance Statement

--
--
-- Generated by BACnet Protocol Stack library EPICS tool
-- BACnet/IP Interface for BACnet-stack Devices
-- http://sourceforge.net/projects/bacnet/
--
--

Vendor Name: "Nube iO Operations Pty Ltd"
Product Name: "Nube iO rubix"
Product Model Number: "rubix"
Product Description: "nube bacnet server"

BIBBs Supported:
```

shomaglasang, 2022-08-06:
Implemented the above mate.
```
PICS 0
BACnet Protocol Implementation Conformance Statement

--
--
-- Generated by BACnet Protocol Stack library EPICS tool
-- BACnet/IP Interface for BACnet-stack Devices
-- http://sourceforge.net/projects/bacnet/
--
--

Vendor Name: "Nube iO Operations Pty Ltd"
Product Name: "Nube iO rubix"
Product Model Number: "rubix"
Product Description: "nube bacnet server"

BIBBs Supported:
{

```

closed 2022-08-06 by shomaglasang