NubeIO/driver-bacnet #35: add bacnet master (bacnet-client)

NubeDev, 2022-12-15:
the idea would be that we could send the bacnet stack a message over MQTT and then MQTT would publish the result back on a topic

we need to resue the same bacnet instance so we can run both services on the same ethenet port

example send a whois

publish to a topic

`bacnet/whois/send` {network:eth0, timeout:2000}

basic scope

- send `whois` `./bacwi` 
- read a device `object list`
- `read present values` or priorty send a list and responsed back with the values [ai1, ai2]
- `write present values` or priorty send a list and responsed back with the values {{point:ao1, pri:16}}

Shiny380, 2022-12-15:
Goal is to be able to run a Server AND a Master at the same time. This enabled 2 devices to have two-way communication with each other, i.e. 2 of our Rubix-Computes.

We will need to do a smaller scale test first to make sure this will work on the same port and that messages get in and out still.
The main requirement is the you can do master requests but the server still has to run and be readable

shomaglasang, 2023-01-04:
@NubeDev @Shiny380 mate so basically the bacnet server subscribes to MQTT topics. And these topics correspond to commands  (e.g. read present values of an object instance, write object name of an instance ). And the response/result is published over MQTT, perhaps in a different topic so the server will not read them back. Am I correct mate? The published value from the server will be in json format.

closed 2023-05-30 by NubeDev