NubeIO/driver-bacnet #36: start bacnet point numbers at 1 not 0

NubeDev, 2022-12-15:
eg: start AO at AO1 not AO0

shomaglasang, 2022-12-20:
@NubeDev mate, the point numbers now start at 1.
Here's sample output for AO:
```
./bin/bacrp 1234 1 0 77 --mac 10.104.0.11:47808
BACnet Error: object: unknown-object
```
```
./bin/bacrp 1234 1 1 77 --mac 10.104.0.11:47808
"AO_1_SPARE"
```
```
./bin/bacrp 1234 1 2 77 --mac 10.104.0.11:47808
"AO_2_SPARE"
```
For AI:
```
./bin/bacrp 1234 0 0 77 --mac 10.104.0.11:47808
BACnet Error: object: unknown-object
```
```
./bin/bacrp 1234 0 1 77 --mac 10.104.0.11:47808
"AI_1_SPARE"
```

closed 2022-12-20 by shomaglasang

NubeDev, 2022-12-20:
on the latest build i can see AO0, AI0 and so on

shomaglasang, 2022-12-21:
just for validation mate, can you pls give me the command you use for AO or AI? In relation to this, I found a bug with the max count, it is minus one when you do a list command. Checking on this atm.

NubeDev, 2022-12-21:
./bacepics 1234

shomaglasang, 2023-01-01:
@NubeDev mate there is still a problem with the indexing. Have to change the common functions but I think it messed up the other parts. I already identified the offending parts. I will wrap this up this week.

shomaglasang, 2023-01-01:
@NubeDev  I will create a new release today mate. The object list now looks like below.
```
    object-list: {
        (device, 1234), (analog-input, 1), (analog-input, 2),
        (analog-output, 1), (analog-output, 2), (analog-value, 1),
        (analog-value, 2), (analog-value, 3), (analog-value, 4),
        (analog-value, 5), (analog-value, 6), (binary-input, 1),
        (binary-input, 2), (binary-input, 3), (binary-input, 4),
        (binary-output, 1), (binary-output, 2), (binary-value, 1),
        (binary-value, 2), (binary-value, 3), (binary-value, 4),
        (binary-value, 5) }
```