NubeIO/driver-bacnet #52: add in token in read/write and whois body

NubeDev, 2023-04-21:
so we can match up the reponses from a cmd

shomaglasang, 2023-04-25:
@NubeDev there's probably more than 1 tokens in the JSON body. The supported tokens can be specified in the config file instead of having them static.

Example:
```
json_tokens: 
  - token1
  - token2
```

shomaglasang, 2023-04-30:
Sample results.
Read Request:
```
mqtt pub -h 104.219.42.42 -p 1883 -t bacnet/cmd/read_value -m '{"objectType":"0","objectInstance":"1","property":"85","deviceInstance":"5678","mac":"10.104.0.11:47900", "txn_source":"abcde", "txn_number":"123"}'
```
Read Reply:
```
bacnet/cmd_result/read_value/ai/1/pv: { "value" : "123.500000" , "deviceInstance" : "5678" , "mac" : "10.104.0.11:47900" , "txn_source" : "abcde", "txn_number" : "123"}
```

Write Request:
```
mqtt pub -h 104.219.42.42 -p 1883 -t bacnet/cmd/write_value -m '{"objectType":"0","objectInstance":"1","property":"85","deviceInstance":"5678","mac":"10.104.0.11:47900", "value":"123.50", "txn_source":"abcd","txn_number":"123"}'
```
Write Reply:
```
bacnet/cmd_result/write_value/ai/1/pv: { "value" : "123.50" , "deviceInstance" : "5678" , "mac" : "10.104.0.11:47900" , "txn_source" : "abcd", "txn_number" : "123"}
```

Whois Request:
```
mqtt pub -h 104.219.42.42 -p 1883 -t bacnet/cmd/whois -m '{"txn_source":"abcd","txn_number":"123"}'
```
Whos Reply:
```
bacnet/cmd_result/whois: { "value" : [  ], "txn_source" : "abcd", "txn_number" : "123" }
```

closed 2023-04-30 by NubeDev