NubeIO/driver-bacnet #53: add index to body

NubeDev, 2023-04-22:
if its empty then ignore

```
{
    "objectType": "0",
    "objectInstance": "1",
    "property": "77",
    "deviceInstance": "2508",
    "mac": "192.168.15.48,
    "index": ""
{
```

shomaglasang, 2023-05-07:
device object list
index 0, returns the number of object instances
request:
```
'{"objectType":"8","objectInstance":"5678","property":"76","deviceInstance":"5678","mac":"10.104.0.11:47900", "index":"0"}'
```
response:
```
bacnet/cmd_result/read_value/device/5678/object_list: { "value" : "22" , "deviceInstance" : "5678" , "mac" : "10.104.0.11:47900" , "index" : "0" }
```

no index, returns the list of object instances
request:
```
'{"objectType":"8","objectInstance":"5678","property":"76","deviceInstance":"5678","mac":"10.104.0.11:47900"}'
```
response:
```
bacnet/cmd_result/read_value/device/5678/object_list: { "value" : [["device", "5679"],["analog-input", "1"],["analog-input", "2"],["analog-output", "1"],["analog-output", "2"],["analog-value", "1"],["analog-value", "2"],["analog-value", "3"],["analog-value", "4"],["analog-value", "5"],["analog-value", "6"],["binary-input", "1"],["binary-input", "2"],["binary-input", "3"],["binary-input", "4"],["binary-output", "1"],["binary-output", "2"],["binary-value", "1"],["binary-value", "2"],["binary-value", "3"],["binary-value", "4"],["binary-value", "5"]] , "deviceInstance" : "5678" , "mac" : "10.104.0.11:47900" }
```

index 1, returns the first object instance
request:
```
'{"objectType":"8","objectInstance":"5678","property":"76","deviceInstance":"5678","mac":"10.104.0.11:47900", "index":"1"}'
```
response:
```
bacnet/cmd_result/read_value/device/5678/object_list: { "value" : ["device", "5679"] , "deviceInstance" : "5678" , "mac" : "10.104.0.11:47900" , "index" : "1" }
```

index 22, returns the last object instance
request:
```
'{"objectType":"8","objectInstance":"5678","property":"76","deviceInstance":"5678","mac":"10.104.0.11:47900", "index":"22"}'
```
response:
```
bacnet/cmd_result/read_value/device/5678/object_list: { "value" : ["binary-value", "5"] , "deviceInstance" : "5678" , "mac" : "10.104.0.11:47900" , "index" : "22" }
```

using timeout, 120 sec
request:
```
'{"objectType":"8","objectInstance":"5678","property":"76","deviceInstance":"5678","mac":"10.104.0.11:47900", "index":"22", "timeout":"120"}'
```

shomaglasang, 2023-05-15:
Index and timeout already supported. Closing this case.

closed 2023-05-15 by shomaglasang