NubeIO/driver-bacnet #57: bacnet-client (bacnet master) add support MSI, MSO, MSV

NubeDev, 2023-05-08:
```
Multistate Input	 13
Multistate Output 14
Multistate Value	 19 
```

shomaglasang, 2023-05-25:
hi mate @NubeDev I will use the following format for the default object name.
MSI_<instance_index>_SPARE
MSO_<instance_index>_SPARE
MSV_<instance_index>_SPARE

ex:
```
MSI_1_SPARE
```
The max number of objects will be configurable from yaml file, e.g.
```
msi_max: 5
mso_max: 10
msv_max: 10
```

NubeDev, 2023-05-25:
We only need them for `master` not `server`

shomaglasang, 2023-05-26:
Do we need to change the indexing? The first object instance starts at 0. Should we change it to 1?

NubeDev, 2023-05-26:
Indexing for what?

We are not doing the server, just master its for reading and writing

shomaglasang, 2023-05-26:
Example to read the present value of the first MSI object, the bacnet command would be
```
bacrp 1234 13 0 85
```
We changed the indexing for AI, AO, etc to start at 1. So this would
```
bacrp 1234 13 1 85
```
Should we change this? So in the MQTT command, the index will be 1 like
```
{"objectType":"13","objectInstance":"1","property":"85","deviceInstance":"5678","mac":"10.104.0.11:47900"}
```

shomaglasang, 2023-05-28:
hi mate @NubeDev I added this in v0.0.45. Can you pls check if you can and pls Let me know if I missed anything. Thanks

closed 2023-05-30 by shomaglasang