NubeIO/driver-bacnet #58: add support for write priority

NubeDev, 2023-05-09:


CanuckMarc, 2023-05-09:
The bacnet `write priority array` command should support writing or releasing any/all priorities with 1 call.  The PATCH body would be like the following: 

```
{
    "objectType": "2",
    "objectInstance": "1",
    "deviceInstance": "2508",
    "mac": "192.168.15.48:47808"
    "priorityArray": { 
        "_1": "12.34",    // set bacnet point priority 1 to 12.34
        "_5": "null",       // release bacnet point priority 5
        "_12": "null",     // release bacnet point priority 12
        "_16": "55",      // set bacnet point priority 16 to 55
    }
}
```
The `priorityArray` property can contain up to 16 key/value pairs (`_1` --> `_16`), the values can be either numbers, or `null`.  It is invalid to have an empty `priorityArray`, it must contain at least one key/value pair.  

When this PATCH call is received, the bacnet server will compare the received values against the existing Priority Array on the BACnet point.  Each priority (`_1` --> `_16`) is then updated, and the BACnet Point Present Value is updated.  A value of `null` will release the existing value on the BACnet point at the same priority level.  

The following is an example PATCH call to release all priority values:
```
{
    "objectType": "2",
    "objectInstance": "1",
    "deviceInstance": "2508",
    "mac": "192.168.15.48:47808"
    "priorityArray": { 
        "_1": "null", 
        "_2": "null", 
        "_3": "null", 
        "_4": "null", 
        "_5": "null", 
        "_6": "null", 
        "_7": "null", 
        "_8": "null", 
        "_9": "null", 
        "_10": "null", 
        "_11": "null", 
        "_12": "null", 
        "_13": "null", 
        "_14": "null", 
        "_15": "null", 
        "_16": "null"
    }
}
```

shomaglasang, 2023-05-09:
Working on this @NubeDev @CanuckMarc

CanuckMarc, 2023-05-14:
@shomaglasang - Status Update?

NubeDev, 2023-05-15:
@CanuckMarc 
I spoke with him he said by the end of this week should be ready

shomaglasang, 2023-05-15:
hi mate @CanuckMarc , still working on this. Should be ready by the end of the week, wednesday the soonest. Will let you know when a new release is out. Thanks

shomaglasang, 2023-05-21:
Hi @CanuckMarc @NubeDev , this is included in v0.0.44.

Sample write to AO priority array in a single request.
Topic: bacnet/cmd/write_value
```
{"objectType":"1","objectInstance":"1","property":"87","deviceInstance":"5678","mac":"192.168.15.17:47900", "value":"100.50", "priorityArray":{"_1":"10.10", "_2":"20.20", "_5":"null", "_16":"100"}}
```
Published response:
```
bacnet/cmd_result/write_value/ao/1/pri: { "value" : {"_1" : "10.10" , "_2" : "20.20" , "_5" : "Null" , "_16" : "100"} , "deviceInstance" : "5678" , "mac" : "192.168.15.17:47900" }
```

Read AO priority array
Topic: bacnet/cmd/read_value
```
{"objectType":"1","objectInstance":"1","property":"87","deviceInstance":"5678","mac":"192.168.15.17:47900"}
```
Published response:
```
bacnet/cmd_result/read_value/ao/1/pri: { "value" : ["10.100000","20.200000","Null","Null","Null","Null","Null","Null","Null","Null","Null","Null","Null","Null","Null","100.000000"] , "deviceInstance" : "5678" , "mac" : "192.168.15.17:47900" }
```

CanuckMarc, 2023-05-22:
@shomaglasang 

I am having issues running this command.  It seems to accept the write command, but doesn't respond with anything.  See my notes below:

Before I get into the stuff that doesn't work, here is some stuff that does work:

These are run using postman to call the FF API functions:
```
POST
URL: localhost:1660/api/plugins/api/bacnetmaster/read/pv
BODY: {
    "objectType": "1",
    "objectInstance": "1",
    "property": "85",
    "deviceInstance": "1000",
    "mac": "192.168.1.10:47808"
}
RESPONSE:
{
    "objectType": "1",
    "objectInstance": "1",
    "deviceInstance": 1000,
    "mac": "192.168.1.10:47808",
    "value": 345,
    "txn_source": "ff",
    "txn_number": "833689"
}
```

```
POST
URL: localhost:1660/api/plugins/api/bacnetmaster/read/pri
BODY: {
    "objectType": "1",
    "objectInstance": "1",
    "property": "87",
    "deviceInstance": "1000",
    "mac": "192.168.1.10:47808"
}
RESPONSE:
{
    "objectType": "1",
    "objectInstance": "1",
    "value": {
        "_1": null,
        "_2": null,
        "_3": null,
        "_4": null,
        "_5": null,
        "_6": null,
        "_7": null,
        "_8": null,
        "_9": null,
        "_10": null,
        "_11": null,
        "_12": null,
        "_13": null,
        "_14": null,
        "_15": null,
        "_16": 345
    },
    "deviceInstance": 1000,
    "mac": "192.168.1.10:47808",
    "txn_source": "ff",
    "txn_number": "514686"
}
```

These next ones are MQTT messages that i'm sending from Node-Red (to keep it simple), they are basically the same as you've written in your comments above:

This read command works fine:
```
msg.topic = "bacnet/cmd/read_value"
msg.payload = {
    "objectType": "1", 
    "objectInstance": "1", 
    "property": "87", 
    "deviceInstance": "1000", 
    "mac": "192.168.1.10:47808", 
}
```
The response:
```
{"topic":"bacnet/cmd_result/read_value/ao/1/pri","payload":{"value":["Null","Null","Null","Null","Null","Null","Null","Null","Null","Null","Null","Null","Null","Null","Null","345.000000"],"deviceInstance":"1000","mac":"192.168.1.10:47808"},"qos":0,"retain":false,"_topic":"bacnet/cmd_result/read_value/ao/1/pri","_msgid":"83577eb012f2c22c"}
```

The write command doesn't work:  No Reponse
```
msg.topic = "bacnet/cmd/write_value"
msg.payload = {
    "objectType": "1", 
    "objectInstance": "1", 
    "property": "87", 
    "deviceInstance": "1000", 
    "mac": "192.168.1.10:47808", 
    "priorityArray": {"_16": "100"}
}
```
When I run the above command, this is what I see on the BACnet Server (v0.0.44):
```
MQTT message arrived: => 159877
- version: [1]
- topic  : [bacnet/cmd/write_value]
- value  : [{"objectType":"1","objectInstance":"1","property":"87","deviceInstance":"1000","mac":"192.168.1.10:47808","priorityArray":{"_16":"100"}}]
MQTT Topic Tokens:
- token[0] => [bacnet]
- token[1] => [cmd]
- token[2] => [write_value]
The json string:

{ "objectType": "1", "objectInstance": "1", "property": "87", "deviceInstance": "1000", "mac": "192.168.1.10:47808", "priorityArray": { "_16": "100" } }

- checking key: [txn_source]
- checking key: [txn_number]
- [0] index: [16] , value: [100]
write request_invoke_id: 7
```

THE BIG ISSUE IS THAT AFTER I RUN THIS WRITE COMMAND THE READ PRIORITY COMMANDS NO LONGER WORK:

Same command that worked earlier no longer works:
```
msg.topic = "bacnet/cmd/read_value"
msg.payload = {
    "objectType": "1", 
    "objectInstance": "1", 
    "property": "87", 
    "deviceInstance": "1000", 
    "mac": "192.168.1.10:47808", 
}
```
NO RESPONSE.  On the bacnet server there is this:
```
MQTT message arrived: => 159877
- version: [1]
- topic  : [bacnet/cmd/read_value]
- value  : [{"objectType":"1","objectInstance":"1","property":"87","deviceInstance":"1000","mac":"192.168.1.10:47808"}]
MQTT Topic Tokens:
- token[0] => [bacnet]
- token[1] => [cmd]
- token[2] => [read_value]
The json string:

{ "objectType": "1", "objectInstance": "1", "property": "87", "deviceInstance": "1000", "mac": "192.168.1.10:47808" }

- checking key: [txn_source]
- checking key: [txn_number]
read request_invoke_id: 8
- Aged request found: 8
```

AND THIS ONE VIA POSTMAN DOESN'T WORK ANYMORE:
```
POST
URL: localhost:1660/api/plugins/api/bacnetmaster/read/pri
BODY: {
    "objectType": "1",
    "objectInstance": "1",
    "property": "87",
    "deviceInstance": "1000",
    "mac": "192.168.1.10:47808"
}
```

To get it working again, I need to restart the device running bacnet server.

shomaglasang, 2023-05-22:
Thanks for the feedback @CanuckMarc , looks like there's  buffer overflow. I will replicate and fix. Will get back asap.

shomaglasang, 2023-05-22:
Hi @CanuckMarc I tried but can't replicate the issue on the office device 192.168.15.17.
Can you pls post the full server stdout messages, after the line below:
```
- [0] index: [16] , value: [100]
```
I want to compare the result on my device. It looks like this on the device I am testing:
```
- checking key: [txn_source]
- checking key: [txn_number]
- [0] index: [16] , value: [100]
BVLC: Send Original-Unicast-NPDU 192.168.15.48:47808
BIP: Sending MPDU-> 192.168.15.48:47808 (26 bytes)
write request_invoke_id: 4
BIP: Received MPDU-> 192.168.15.48:47808 (9 bytes)
BVLC: Received BVLC (BBMD Enabled) 192.168.15.48:47808
BVLC: Received Original-Unicast-NPDU 192.168.15.48:47808
BVLC: *** my_addr: 192.168.15.17:47808
BVLC: *** addr: 192.168.15.48:47808
BVLC: Original-Unicast-NPDU NPDU=MTU[4] len=5
BIP: Received NPDU-> 192.168.15.48:47808 (5 bytes)
- bacnet_client_write_value_handler() => 23248
-- invoke_id: 4
-- Request with pending reply found!
- request with ID: 4 found for deleting
- write value result topic: bacnet/cmd_result/write_value/ao/1/pri
- write value result topic value: { "value" : {"_16" : "100"} , "deviceInstance" : "2508" , "mac" : "192.168.15.48:47808" }
MQTT published topic: "bacnet/cmd_result/write_value/ao/1/pri"
Message with token value 0 delivery confirmed
```

CanuckMarc, 2023-05-23:
@shomaglasang That seems to be working.  I think maybe my issue was an old BACnet Server that wasn't dealing with the requests properly.  I'll let you know if I run into any other issues.

shomaglasang, 2023-05-23:
@CanuckMarc yes that was I though. Please let me know if you encounter any issue. I will check right away. I will continue working on https://github.com/NubeIO/driver-bacnet/issues/57. thanks mate.

closed 2023-05-23 by shomaglasang

NubeDev, 2024-05-30:
@shomaglasang 

I think there is a bug in driver-bacnet for writring the priorityArray

```
{
    "objectType": "1",
    "objectInstance": "1",
    "property": "87",
    "deviceInstance": "1",
    "mac": "192.168.15.140:47808",
    "dnet": "0",
    "dadr": "0",
    "priorityArray": {
        "_16": 666
    },
    "txn_source": "ff",
    "txn_number": "d170c63b2e10",
    "timeout": 2
}
```
see response

```json
{"error":"write-access-denied","objectType":"1","objectInstance":"1","property":"87","index":"16","priority":"0","deviceInstance":"1","mac":"192.168.15.140:47808","dadr":"0","dnet":"0","priorityArray":{"_16":"666"},"txn_source":"ff","txn_number":"d170c63b2e10"}
```