NubeIO/driver-bacnet #70: BACnet master add support for multiple write

NubeDev, 2023-08-17:


shomaglasang, 2023-09-12:
@NubeDev Hi mate, just want your confirmation. This would be the bacwrp, right?
Example:
```
bacwpm 123 1 44 85 16 4 100 1 45 85 16 4 100
```

shomaglasang, 2023-09-17:
@NubeDev mate should we also support for multiple read? The link to the source file above is actually for read properly multiple.

shomaglasang, 2023-10-01:
Hi mate @NubeDev , I also supported read multiple property. Please test the latest version v0.0.50 when you have time mate. Thanks.

Here's a sample request payload.
```
{"deviceInstance":"5678", "objects": [{"objectType":"0","objectInstance":"1","property":"117"}, {"objectType":"1","objectInstance":"1","property":"85"}, {"objectType":"4","objectInstance":"1","property":"87"}], "mac":"10.104.0.11:47900"}
```

Sample result:
```
{ "deviceInstance" : "5678", "objects" : [{"objectType" : "0", "objectInstance" : "1", "property" : "117", "value" : "no-units"},{"objectType" : "1", "objectInstance" : "1", "property" : "85", "value" : "0.000000"},{"objectType" : "4", "objectInstance" : "1", "property" : "87", "value" : ["Null","Null","Null","Null","Null","Null","Null","Null","Null","Null","Null","Null","Null","Null","Null","Null"]}], "mac" : "10.104.0.11:47900"  }
```

shomaglasang, 2023-11-07:
Closing this issue.

closed 2023-11-07 by shomaglasang