NubeIO/driver-bacnet #85: reduce the amout of bacnet req on PICs

NubeDev, 2023-11-24:
We are finding a lot of BACnet devices can be slow in the real world so we want to look at reducing the size of the PIC call

## device request
keep all the device `requests` as is

## points request

on request the following `name`, `id`, `type, units (if supported)`, `description`

  { 
    **object-identifier**: (analog-input, 1)
    **object-name**: "AI_1_SPARE"
    **object-type**: analog-input
    present-value: ?
    status-flags: {false,false,false,false}
    event-state: normal
    out-of-service: FALSE
    **units**: no-units
    **description**: "AI_1_SPARE"
    reliability: ?
    cov-increment: 1.000000
    time-delay: 170930739
    notification-class: 4194303
    high-limit: 4122011038318277427200.000000
    low-limit: 17749878618931926680750522368.000000
    deadband: 0.000000
    limit-enable: {false,false}
    event-enable: {false,false,false}
    acked-transitions: {true,true,true}
    notify-type: 1
    event-time-stamps: { 2155-255-255T255:255:255.255,2155-255-255T255:255:255.255,2155-255-255T255:255:255.255 }
    -- proprietary 9997: 90.510002
    -- proprietary 9998: 90
    -- proprietary 9999: -200
  }, 

## only call points so dont call `alarms`, `schedules`, `trends`

shomaglasang, 2023-11-29:
@NubeDev currently testing v0.0.59.
Here's a sample point objects:
```
{
"object-identifier" : ["analog-input", 1],
"object-name" : "AI_1_SPARE",
"object-type" : "analog-input",
"units" : "no-units"},
{
"object-identifier" : ["binary-value", 20],
"object-name" : "BV_20_SPARE",
"object-type" : "binary-value",
"description" : "BV_20_SPARE"},
```

Properties in Device Object is as is, except in the "object-list" only supported objects are included.
```
{
"object-identifier" : ["device", 17],
"object-name" : "Nube IO- DEV 17",
"object-type" : "device",
"system-status" : "operational",
"vendor-name" : "Nube iO Operations Pty Ltd",
"vendor-identifier" : "1173",
"model-name" : "rubix",
"firmware-revision" : "1.0.0",
"application-software-version" : "1.0",
"protocol-version" : "1",
"protocol-revision" : "17",
"protocol-services-supported" : ["T","F","F","T","F","T","T","T","F","F","F","F","T","F","T","T","T","T","F","F","T","F","F","F","F","F","F","F","T","F","T","F","T","T","T","T","T","F","F","T","F"],
"protocol-object-types-supported" : ["T","T","T","T","T","T","F","F","T","F","F","F","F","T","T","F","F","F","F","T","F","F","F","F","F","F","F","F","F","F","F","F","F","F","F","F","F","F","F","F","F","F","F","F","F","F","F","F","F","F","F","F","F","F","F","F","F"],
"object-list" : [["device", 18],["analog-input", 1],["analog-input", 2],["analog-input", 3],["analog-input", 4],["analog-input", 5],["analog-input", 6],["analog-output", 1],["analog-output", 2],["analog-value", 1],["analog-value", 2],["analog-value", 3],["analog-value", 4],["analog-value", 5],["analog-value", 6],["analog-value", 7],["analog-value", 8],["analog-value", 9],["analog-value", 10],["binary-input", 1],["binary-input", 2],["binary-input", 3],["binary-input", 4],["binary-input", 5],["binary-input", 6],["binary-input", 7],["binary-input", 8],["binary-input", 9],["binary-input", 10],["binary-output", 1],["binary-output", 2],["binary-output", 3],["binary-output", 4],["binary-output", 5],["binary-output", 6],["binary-output", 7],["binary-output", 8],["binary-output", 9],["binary-output", 10],["binary-value", 1],["binary-value", 2],["binary-value", 3],["binary-value", 4],["binary-value", 5],["binary-value", 6],["binary-value", 7],["binary-value", 8],["binary-value", 9],["binary-value", 10],["binary-value", 11],["binary-value", 12],["binary-value", 13],["binary-value", 14],["binary-value", 15],["binary-value", 16],["binary-value", 17],["binary-value", 18],["binary-value", 19],["binary-value", 20],["multi-state-input", 1],["multi-state-input", 2],["multi-state-input", 3],["multi-state-input", 4],["multi-state-output", 1],["multi-state-output", 2],["multi-state-output", 3],["multi-state-output", 4],["multi-state-value", 1],["multi-state-value", 2],["multi-state-value", 3],["multi-state-value", 4]],
"max-apdu-length-accepted" : "1476",
"segmentation-supported" : "no-segmentation",
"apdu-timeout" : "3000",
"number-of-APDU-retries" : "3",
"device-address-binding" : "?",
"database-revision" : "?",
"proprietary 4194338" : "Nube iO rubix",
"description" : "nube bacnet server",
"local-time" : "?",
"utc-offset" : "?",
"local-date" : "?",
"daylight-savings-status" : "?",
"location" : "USA",
"active-cov-subscriptions" : "Null",
"time-synchronization-recipients" : "Null",
"time-synchronization-interval" : "0",
"align-intervals" : "FALSE",
"interval-offset" : "0"}
```

shomaglasang, 2023-12-14:
Hi @NubeDev  mate, this is supported already. Please let me know if there are other features or enhancements to be worked on Thanks.

closed 2023-12-14 by shomaglasang