NubeIO/driver-bacnet #95: MQTT payload has changed!!!

CanuckMarc, 2024-02-20:
@shomaglasang 

The MQTT payload of the `bacnet/av/1/pri` topic has changed between `v0.0.64` and `v0.0.65`.  

Please explain why, as it may affect the operation of our system components (ROS and Wires).  

Being that you are making changes to `driver-bacnet` you MUST keep a log of the changes in the CHANGELOG.md file.  Please ensure this gets updated.  

Please also update the README.md file so that it clearly describes how to run the Release versions of `driver-bacnet`.  This includes how to set up the config file, and the start commands.

shomaglasang, 2024-02-20:
@CanuckMarc noted mate.
The payload for priority array for ao,av,bo and bv was changed in v0.0.65. Wasn't aware this is being used. I changed the payload to make it json compliant so it is easier to read. This is needed for persistent values restoration when bacnet-server starts up (can be disabled from config).

From this format
```
{ "value" : "{123.500000,Null,Null,Null,Null,Null,Null,Null,Null,300.000000,Null,Null,Null,Null,Null,222.300003}" , "uuid" : "" }
```
into
```
{ "value" : ["123.500000","Null","Null","Null","Null","Null","Null","Null","Null","300.000000","Null","Null","Null","Null","Null","222.300003"] , "uuid" : "" }
```

CanuckMarc, 2024-02-20:
@shomaglasang chat with @Shiny380 as there are a bunch of services that use the `driver-bacnet` MQTT.  Need to coordinate so we don't break stuff.

closed 2024-02-29 by Shiny380