NubeIO/driver-bacnet #104: double publishing only over MQTT

NubeDev, 2024-05-31:
It seems driver bacnet is sending the same request twice every time

[image]

[image]

shomaglasang, 2024-07-23:
Hi mate @NubeDev checking this one.

shomaglasang, 2024-07-28:
Hi @NubeDev I can't seem to replicate this on my devel environment and in the office device at 192.168.15.17. From your screenshot, It looks like the driver bacnet received the same (duplicate) MQTT message twice. Am I correct? The topic of the duplicate messages from your screenshot is "bacnet/cmd/write_value". These messages (topic) are not published by the driver bacnet but by a client over MQTT.

shomaglasang, 2024-08-10:
Hi mate @NubeDev The fix is included in 1.0.2-rc.3 https://github.com/NubeIO/driver-bacnet/releases/tag/v1.0.2-rc.3
Please try this latest version. Initial testing on the office device is good. I will merge the PR if I don't find any issue. Thanks

shomaglasang, 2024-08-18:
Mate @NubeDev Closing this issue.

closed 2024-08-18 by shomaglasang