NubeIO/driver-bacnet #122: Discover device data discrepancy over the time

RaiBnod, 2024-06-20:
When we are having this issue https://github.com/NubeIO/driver-bacnet/issues/121, the device gets discovered **(but names don't look good)** but points don't (`0 points`).

### When it's having that issue:

[image]

### After `driver-bacnet` restart:

[image]

RaiBnod, 2024-06-20:
It falls on the same issue: https://github.com/NubeIO/driver-bacnet/issues/121

The `whois` command is working fine, names are not being returned. So, it's just adding `deviceID_` in the prefix. So, I expect it should work if we make this https://github.com/NubeIO/driver-bacnet/issues/121 work.

shomaglasang, 2024-06-24:
@RaiBnod @NubeDev 
This should be resolved by the fix for [#121](https://github.com/NubeIO/driver-bacnet/issues/121)

closed 2024-06-24 by RaiBnod