NubeIO/driver-bacnet #125: Incorrect JSON while reading priority array Null

RaiBnod, 2024-07-03:
Parent ticket: https://github.com/NubeIO/module-core-bacnetmaster/issues/41

When the BACnet server's priority array is null, this code can produce an incorrect JSON structure.

It currently exists in the type `mso`, check others as well.

### An example of an incorrect response:

```JSON
{ "value" : Null , "deviceInstance" : "3" , "dnet" : "0" , "dadr" : "0" , "mac" : "192.168.0.3:47808" , "txn_source" : "ff", "txn_number" : "b92f5ffd5f64"}
```

### The correct response would have been:

```JSON
{ "value" : ["Null","Null","Null","Null","Null","Null","Null","Null","Null","Null","Null","Null","Null","Null","Null","1"] , "deviceInstance" : "3" , "dnet" : "0" , "dadr" : "0" , "mac" : "192.168.0.3:47808" , "txn_source" : "ff", "txn_number" : "93951ba0befb"}
```

### More logs:

```bash
pi@NubeIO-RC-5:~ $ mosquitto_sub -t 'bacnet/#' -v
bacnet/program/2508/state { "value" : "started" , "uuid" : "" }
bacnet/program/54321/state { "value" : "started" , "uuid" : "" }
bacnet/cmd/read_value {"objectType":"4","objectInstance":"65550","property":"87","deviceInstance":"3","mac":"192.168.0.3:47808","dnet":"0","dadr":"0","txn_source":"ff","txn_number":"93951ba0befb","timeout":10,"known_object_list":null}
bacnet/cmd_result/read_value/bo/65550/pri { "value" : ["Null","Null","Null","Null","Null","Null","Null","Null","Null","Null","Null","Null","Null","Null","Null","1"] , "deviceInstance" : "3" , "dnet" : "0" , "dadr" : "0" , "mac" : "192.168.0.3:47808" , "txn_source" : "ff", "txn_number" : "93951ba0befb"}
bacnet/cmd/read_value {"objectType":"13","objectInstance":"4","property":"85","deviceInstance":"3","mac":"192.168.0.3:47808","dnet":"0","dadr":"0","txn_source":"ff","txn_number":"42ebf805d98e","timeout":10,"known_object_list":null}
bacnet/cmd_result/read_value/msi/4/pv { "value" : "255" , "deviceInstance" : "3" , "dnet" : "0" , "dadr" : "0" , "mac" : "192.168.0.3:47808" , "txn_source" : "ff", "txn_number" : "42ebf805d98e"}
bacnet/cmd/read_value {"objectType":"13","objectInstance":"6","property":"85","deviceInstance":"3","mac":"192.168.0.3:47808","dnet":"0","dadr":"0","txn_source":"ff","txn_number":"fea715acd6f9","timeout":10,"known_object_list":null}
bacnet/cmd_result/read_value/msi/6/pv { "value" : "1" , "deviceInstance" : "3" , "dnet" : "0" , "dadr" : "0" , "mac" : "192.168.0.3:47808" , "txn_source" : "ff", "txn_number" : "fea715acd6f9"}
bacnet/cmd/read_value {"objectType":"13","objectInstance":"8","property":"85","deviceInstance":"3","mac":"192.168.0.3:47808","dnet":"0","dadr":"0","txn_source":"ff","txn_number":"d2ccd5493eb8","timeout":10,"known_object_list":null}
bacnet/cmd_result/read_value/msi/8/pv { "value" : "2" , "deviceInstance" : "3" , "dnet" : "0" , "dadr" : "0" , "mac" : "192.168.0.3:47808" , "txn_source" : "ff", "txn_number" : "d2ccd5493eb8"}
bacnet/cmd/read_value {"objectType":"13","objectInstance":"32772","property":"85","deviceInstance":"3","mac":"192.168.0.3:47808","dnet":"0","dadr":"0","txn_source":"ff","txn_number":"83db71666e63","timeout":10,"known_object_list":null}
bacnet/cmd_result/read_value/msi/32772/pv { "value" : "255" , "deviceInstance" : "3" , "dnet" : "0" , "dadr" : "0" , "mac" : "192.168.0.3:47808" , "txn_source" : "ff", "txn_number" : "83db71666e63"}
bacnet/cmd/read_value {"objectType":"13","objectInstance":"32774","property":"85","deviceInstance":"3","mac":"192.168.0.3:47808","dnet":"0","dadr":"0","txn_source":"ff","txn_number":"83aaa9de88e3","timeout":10,"known_object_list":null}
bacnet/cmd_result/read_value/msi/32774/pv { "value" : "1" , "deviceInstance" : "3" , "dnet" : "0" , "dadr" : "0" , "mac" : "192.168.0.3:47808" , "txn_source" : "ff", "txn_number" : "83aaa9de88e3"}
bacnet/cmd/read_value {"objectType":"13","objectInstance":"32776","property":"85","deviceInstance":"3","mac":"192.168.0.3:47808","dnet":"0","dadr":"0","txn_source":"ff","txn_number":"eded725b6fad","timeout":10,"known_object_list":null}
bacnet/cmd_result/read_value/msi/32776/pv { "value" : "3" , "deviceInstance" : "3" , "dnet" : "0" , "dadr" : "0" , "mac" : "192.168.0.3:47808" , "txn_source" : "ff", "txn_number" : "eded725b6fad"}
bacnet/cmd/read_value {"objectType":"13","objectInstance":"65540","property":"85","deviceInstance":"3","mac":"192.168.0.3:47808","dnet":"0","dadr":"0","txn_source":"ff","txn_number":"0fdc5ee5a295","timeout":10,"known_object_list":null}
bacnet/cmd_result/read_value/msi/65540/pv { "value" : "255" , "deviceInstance" : "3" , "dnet" : "0" , "dadr" : "0" , "mac" : "192.168.0.3:47808" , "txn_source" : "ff", "txn_number" : "0fdc5ee5a295"}
bacnet/cmd/read_value {"objectType":"13","objectInstance":"65542","property":"85","deviceInstance":"3","mac":"192.168.0.3:47808","dnet":"0","dadr":"0","txn_source":"ff","txn_number":"1711b30e3f28","timeout":10,"known_object_list":null}
bacnet/cmd_result/read_value/msi/65542/pv { "value" : "1" , "deviceInstance" : "3" , "dnet" : "0" , "dadr" : "0" , "mac" : "192.168.0.3:47808" , "txn_source" : "ff", "txn_number" : "1711b30e3f28"}
bacnet/cmd/read_value {"objectType":"13","objectInstance":"65544","property":"85","deviceInstance":"3","mac":"192.168.0.3:47808","dnet":"0","dadr":"0","txn_source":"ff","txn_number":"24dd5a339781","timeout":10,"known_object_list":null}
bacnet/cmd_result/read_value/msi/65544/pv { "value" : "2" , "deviceInstance" : "3" , "dnet" : "0" , "dadr" : "0" , "mac" : "192.168.0.3:47808" , "txn_source" : "ff", "txn_number" : "24dd5a339781"}
bacnet/cmd/read_value {"objectType":"14","objectInstance":"5","property":"87","deviceInstance":"3","mac":"192.168.0.3:47808","dnet":"0","dadr":"0","txn_source":"ff","txn_number":"040271e2f2f3","timeout":10,"known_object_list":null}
bacnet/cmd_result/read_value/mso/5/pri { "value" : Null , "deviceInstance" : "3" , "dnet" : "0" , "dadr" : "0" , "mac" : "192.168.0.3:47808" , "txn_source" : "ff", "txn_number" : "040271e2f2f3"}
bacnet/cmd/read_value {"objectType":"14","objectInstance":"7","property":"87","deviceInstance":"3","mac":"192.168.0.3:47808","dnet":"0","dadr":"0","txn_source":"ff","txn_number":"b8ba52a9dc7b","timeout":10,"known_object_list":null}
bacnet/cmd_result/read_value/mso/7/pri { "value" : Null , "deviceInstance" : "3" , "dnet" : "0" , "dadr" : "0" , "mac" : "192.168.0.3:47808" , "txn_source" : "ff", "txn_number" : "b8ba52a9dc7b"}
bacnet/cmd/read_value {"objectType":"14","objectInstance":"32773","property":"87","deviceInstance":"3","mac":"192.168.0.3:47808","dnet":"0","dadr":"0","txn_source":"ff","txn_number":"36f57eb61277","timeout":10,"known_object_list":null}
bacnet/cmd_result/read_value/mso/32773/pri { "value" : Null , "deviceInstance" : "3" , "dnet" : "0" , "dadr" : "0" , "mac" : "192.168.0.3:47808" , "txn_source" : "ff", "txn_number" : "36f57eb61277"}
bacnet/cmd/read_value {"objectType":"14","objectInstance":"32775","property":"87","deviceInstance":"3","mac":"192.168.0.3:47808","dnet":"0","dadr":"0","txn_source":"ff","txn_number":"e9dba6658b68","timeout":10,"known_object_list":null}
bacnet/cmd_result/read_value/mso/32775/pri { "value" : Null , "deviceInstance" : "3" , "dnet" : "0" , "dadr" : "0" , "mac" : "192.168.0.3:47808" , "txn_source" : "ff", "txn_number" : "e9dba6658b68"}
bacnet/cmd/read_value {"objectType":"14","objectInstance":"65541","property":"87","deviceInstance":"3","mac":"192.168.0.3:47808","dnet":"0","dadr":"0","txn_source":"ff","txn_number":"c5cdca10397a","timeout":10,"known_object_list":null}
bacnet/cmd_result/read_value/mso/65541/pri { "value" : Null , "deviceInstance" : "3" , "dnet" : "0" , "dadr" : "0" , "mac" : "192.168.0.3:47808" , "txn_source" : "ff", "txn_number" : "c5cdca10397a"}
bacnet/cmd/read_value {"objectType":"14","objectInstance":"65543","property":"87","deviceInstance":"3","mac":"192.168.0.3:47808","dnet":"0","dadr":"0","txn_source":"ff","txn_number":"b1af4f7e953a","timeout":10,"known_object_list":null}
bacnet/cmd_result/read_value/mso/65543/pri { "value" : Null , "deviceInstance" : "3" , "dnet" : "0" , "dadr" : "0" , "mac" : "192.168.0.3:47808" , "txn_source" : "ff", "txn_number" : "b1af4f7e953a"}
bacnet/cmd/read_value {"objectType":"0","objectInstance":"32777","property":"85","deviceInstance":"3","mac":"192.168.0.3:47808","dnet":"0","dadr":"0","txn_source":"ff","txn_number":"210024f70725","timeout":10,"known_object_list":null}
bacnet/cmd_result/read_value/ai/32777/pv { "value" : "20.000000" , "deviceInstance" : "3" , "dnet" : "0" , "dadr" : "0" , "mac" : "192.168.0.3:47808" , "txn_source" : "ff", "txn_number" : "210024f70725"}
[... cut ...]
bacnet/cmd/read_value {"objectType":"1","objectInstance":"10","property":"85","deviceInstance":"3","mac":"192.168.0.3:47808","dnet":"0","dadr":"0","txn_source":"ff","txn_number":"a4cfb2db5bc2","timeout":1,"known_object_list":null}
bacnet/cmd_result/read_value/ao/10/pv { "value" : "23.000000" , "deviceInstance" : "3" , "dnet" : "0" , "dadr" : "0" , "mac" : "192.168.0.3:47808" , "txn_source" : "ff", "txn_number" : "a4cfb2db5bc2"}
```

Also, can see the inconsistency of using `null`, `"null"` & `"Null"`, but leave it for now coz if we correct it, it breaks other implementations.

shomaglasang, 2024-07-04:
Thanks @RaiBnod , will check this. So we're using MS objects now. Probably it was not supported yet. Let me know if you're seeing this with Analog and Binary objects.

CanuckMarc, 2024-07-05:
@shomaglasang can you clarify your response?  do we support MultiState BACnet objects?  If we don't currently, then we need to move forward with implementing them ASAP. 

@Shiny380 @NubeDev FYI this will be an issue for VRF systems on CBA.

shomaglasang, 2024-07-05:
@CanuckMarc sorry my statement above is unclear. MultiState objects are supported in driver-bacnet / bacnet-server.

I'm checking on the issue with priority array where the json in the response is incorrect.

shomaglasang, 2024-07-11:
@RaiBnod @CanuckMarc I'm doing testing now. There are other changes needed to Multistate objects to function like Analog and Binary objects (aside from the fix raised in this case).

CanuckMarc, 2024-07-11:
@shomaglasang can you please note the differences here.  The intent is to have the Multistate objects just act like numeric points, just like the Analog and Binary objects.  I know there may be other factors to the Multistate objects, but we may not need to use them.  So ya, please post your plan (and any known issues) here so that I can review. Thanks

shomaglasang, 2024-07-12:
@CanuckMarc the changes needed for multistate (from the stock implementation) so they function like Analog and Binary objects are as follows:

1. Maximum limit of objects can be set from the config file (default value is 4 if not specified).
2. Change start of object index from 0 to 1.
3. Make at least the point value, point name and priority value (if applicable) writable.
4. Support for persistent storage (restore upon startup).

Let me know if I missed anything. Thanks
@NubeDev

CanuckMarc, 2024-07-15:
@shomaglasang ok, can you please proceed with making and testing the required updates for Multi-State BACnet points to work correctly.

Shiny380, 2024-07-16:
@CanuckMarc can you please split this into another ticker for the multi state

CanuckMarc, 2024-07-16:
@Shiny380 done, it's here: https://github.com/NubeIO/driver-bacnet/issues/127.  Can you please assign it to Ronny.

CanuckMarc, 2024-07-16:
@shomaglasang Please provide an update on this JSON issue.

shomaglasang, 2024-07-17:
@CanuckMarc there is a fix already. Will include in the next release, for all multi-state fixes/changes. v1.0.2-rc.2

shomaglasang, 2024-07-20:
@RaiBnod @CanuckMarc the fix is included in 1.0.2-rc.2
Prio array with some values
```
bacnet/cmd_result/read_value/mso/1/pri { "value" : ["1","Null","Null","Null","Null","Null","Null","Null","Null","Null","Null","Null","Null","Null","Null","10"] , "deviceInstance" : "17" , "dnet" : "0" , "dadr" : "0" , "mac" : "192.168.15.17:47808" , "txn_source" : "ff", "txn_number" : "23b00e85d1fb-test"}
```

Prio array without values
```
bacnet/cmd_result/read_value/mso/2/pri { "value" : ["Null","Null","Null","Null","Null","Null","Null","Null","Null","Null","Null","Null","Null","Null","Null","Null"] , "deviceInstance" : "17" , "dnet" : "0" , "dadr" : "0" , "mac" : "192.168.15.17:47808" , "txn_source" : "ff", "txn_number" : "23b00e85d1fb-test"}
```

shomaglasang, 2024-07-23:
Hi @RaiBnod the fix is included in v1.0.2-rc.2. Would be great if you can test and verify on another device. I tested the fix on the office device at 192.168.15.17. Thanks