NubeIO/driver-bacnet #127: Support Multi State Points

CanuckMarc, 2024-07-16:
@shomaglasang  Please use this issue for tracking the BACnet multi state point support.  This is a HIGH priority taks. 

Previous correspondance is on this other issue: https://github.com/NubeIO/driver-bacnet/issues/125

Could you please provide daily updates on your progress and also advise an ETA for the completion of this functionality (including testing).

shomaglasang, 2024-07-17:
ETA for 1.0.2-rc.2 is july 19th

shomaglasang, 2024-07-20:
@CanuckMarc @RaiBnod fix included in 1.0.2-rc.2. Test on office device 192.168.15.17.

shomaglasang, 2024-07-23:
Hi @CanuckMarc will close this issue. The fix is included in 1.0.2-rc.2. I did test it on the office device at 192.168.15.17 and seems to be working fine. Thanks

closed 2024-07-23 by shomaglasang

CanuckMarc, 2024-08-02:
I have reviewed this update and it looks to be working as intended.  Thanks @shomaglasang

shomaglasang, 2024-08-02:
Thanks for your feedback @CanuckMarc