NubeIO/driver-bacnet #131: Device properties

NubeDev, 2024-10-14:
Device properties showing errors

https://sourceforge.net/projects/yetanotherbacnetexplorer/

Using YABE
[image]

And Niagara
[image]

NubeDev, 2024-10-14:
the issue maybe related to as we dont seem to support

[image]

for example my deviceID is `15210`
```
./bacrp 15210 8 15210 8
```

shomaglasang, 2024-10-17:
Mate @NubeDev , the property PROP_ALL (8) is not supported yet.
Do you know the exact type/attribute of this property? Is this like an object list PROP_OBJECT_LIST?

NubeDev, 2024-10-18:
Like i said you need to support like in `bacnet-stack/bin/bacserv` 

```
user@user:~/code/bacnet/bacnet-stack/bin$ ./bacserv 1234
BACnet Server Demo
BACnet Stack Version 1.3.8
BACnet Device ID: 1234
Max APDU: 1476

```

shomaglasang, 2024-10-20:
@NubeDev The output above is a plain string (with new lines). I will implement it with the type "string". I tried to search the web if someone has implemented it but can't find any.

shomaglasang, 2024-10-29:
mate @NubeDev , closing this issue.
Fix is included in 1.0.2-rc.4
https://github.com/NubeIO/driver-bacnet/releases/tag/v1.0.2-rc.4

closed 2024-10-29 by shomaglasang