NubeIO/driver-bacnet #136: Support for bulk read/write

RaiBnod, 2025-03-27:
## RPM [Read Property Multiple]
https://sourceforge.net/p/bacnet/src/ci/master/tree/src/bacnet/rpm.c

## WPM [Write Property Multiple]
https://sourceforge.net/p/bacnet/src/ci/master/tree/src/bacnet/wpm.c

shomaglasang, 2025-03-29:
Hi @RaiBnod @NubeDev Is this reading/writing of multiple properties in a single MQTT bacnet command?
The command line utility programs are `bacrpm` or `bacwpm`.

**bacrpm**
`Usage: bacrpm device-instance object-type object-instance property[index][,property[index]] [object-type ...]
       [--version][--help]`

**bacwpm**
`Usage: bacwpm device-instance object-type object-instance property[index] priority tag value [property[index] priority tag value]
       [--version][--help]`

NubeDev, 2025-03-31:
RPM, WPM
@shomaglasang can you please make these work over MQTT

shomaglasang, 2025-03-31:
@NubeDev @RaiBnod  RPM and WPM over MQTT should work now. Is there a device in the office or somewhere that I can use to test? I will send the format of the MQTT command and the sample output.

RaiBnod, 2025-04-01:
@shomaglasang You can test it on an office device `192.168.15.17`.

shomaglasang, 2025-04-01:
Copy @RaiBnod . I am traveling at the moment. Should be settled tomorrow morning. I'll get back to you asap. Thanks.

shomaglasang, 2025-04-02:
@RaiBnod @NubeDev 
Here are the formats of the MQTT command and response for RPM.
**RPM**
Request:
```
Topic: bacnet/cmd/read_value
JSON Payload: {"deviceInstance":"device_instance", "objects": [{"objectType":"object1_type","objectInstance":"object1_instance","property":"object1_property"}, {"objectType":"object2_type","objectInstance":"object2_instance","property":"object2_property"}, ... ], "mac":"IP:PORT"}
```
Response:
```
Topic: bacnet/cmd_result/read_multiple_value
JSON payload: { "deviceInstance" : "device_instance", "objects" : [{"objectType" : "object1_type", "objectInstance" : "object1_instance", "property" : "object1_property", "value" : "object1_value"}, {"objectType" : "object2_type", "objectInstance" : "object2_instance", "property" : "object2_property", "value" : "object2_value"}, ... ], "mac" : "IP:PORT"  }
```

Example MQTT commands using _mosquitto_pub_:
Request
```
mosquitto_pub -t bacnet/cmd/read_value -m '{"deviceInstance":"15211", "objects": [{"objectType":"2","objectInstance":"1","property":"85"}, {"objectType":"2","objectInstance":"2","property":"85"}, {"objectType":"2","objectInstance":"3","property":"85"}, {"objectType":"5","objectInstance":"1","property":"85"}, {"objectType":"5","objectInstance":"2","property":"85"}], "mac":"192.168.15.211:47808"}'
```

Response:
```
bacnet/cmd_result/read_multiple_value { "deviceInstance" : "15211", "objects" : [{"objectType" : "2", "objectInstance" : "1", "property" : "85", "value" : "0.000000"},{"objectType" : "2", "objectInstance" : "2", "property" : "85", "value" : "0.000000"},{"objectType" : "2", "objectInstance" : "3", "property" : "85", "value" : "0.000000"},{"objectType" : "5", "objectInstance" : "1", "property" : "85", "value" : "inactive"},{"objectType" : "5", "objectInstance" : "2", "property" : "85", "value" : "inactive"}], "mac" : "192.168.15.211:47808"  }
```

shomaglasang, 2025-04-02:
@RaiBnod @NubeDev 
For WPM.
Request
```
Topic: bacnet/cmd/write_value
JSON payload: {"deviceInstance":"device_instance", "objects": [{"objectType":"object1_type","objectInstance":"object1_instance","property":"object1_property", "value":"object1_property_value"}, {"objectType":"object2_type","objectInstance":"object2_instance","property":"object2_property", "value":"object2_property_vlaue"}, ...], "mac":"IP:PORT"}
```

Example MQTT command using mosquitto_pub:
Request
```
mqtt pub -h 104.219.42.42 -p 1883 -t bacnet/cmd/write_value -m '{"deviceInstance":"2530", "objects": [{"objectType":"0","objectInstance":"1","property":"85", "value":"20.130"}, {"objectType":"0","objectInstance":"1","property":"77", "value":"test 111"}, {"objectType":"0","objectInstance":"2","property":"85", "value":"50.12"}, {"objectType":"0","objectInstance":"2","property":"77", "value":"xxx 130"}], "mac":"10.104.0.11:47900"}'
```

Which device in the office that I can write values? I didn't perform any write as others might be using them or running a test. The command above is from the local environment.

RaiBnod, 2025-04-03:
Thanks @shomaglasang for your response. Do you have a complete request/response body structure? Our earlier requests example used to have:

```go
type ReadBody struct {
	ObjectType       string   `json:"objectType"`     // 1 analogue output
	ObjectInstance   string   `json:"objectInstance"` // 1
	Property         string   `json:"property"`       // 85 presetValue
	DeviceInstance   string   `json:"deviceInstance"` // 2508
	Mac              string   `json:"mac"`            // 192.168.15.10:47808
	NetworkNumber    string   `json:"dnet"`           // 1
	MacMSTP          string   `json:"dadr"`           // 2
	Index            string   `json:"index,omitempty"`
	TxnSource        string   `json:"txn_source"`
	TxnNumber        string   `json:"txn_number"`
	ApiTimeout       int      `json:"api_timeout,omitempty"`
	Timeout          int      `json:"timeout,omitempty"`
	KnownObjectsList []object `json:"known_object_list"`
}

type WritePVBody struct {
	ObjectType     string `json:"objectType"`     // 1 analogue output
	ObjectInstance string `json:"objectInstance"` // 1
	Property       string `json:"property"`       // 85 presetValue
	DeviceInstance string `json:"deviceInstance"` // 2508
	Mac            string `json:"mac"`            // 192.168.15.10:47808
	NetworkNumber  string `json:"dnet"`           // 1
	MacMSTP        string `json:"dadr"`           // 2
	Value          string `json:"value"`
	TxnSource      string `json:"txn_source"`
	TxnNumber      string `json:"txn_number"`
	ApiTimeout     int    `json:"api_timeout,omitempty"`
	Timeout        int    `json:"timeout,omitempty"`
}

type WritePriorityBody struct {
	ObjectType     string          `json:"objectType"`     // 1 analogue output
	ObjectInstance string          `json:"objectInstance"` // 1
	Property       string          `json:"property"`       // 85 presetValue
	DeviceInstance string          `json:"deviceInstance"` // 2508
	Mac            string          `json:"mac"`            // 192.168.15.10:47808
	NetworkNumber  string          `json:"dnet"`           // 1
	MacMSTP        string          `json:"dadr"`           // 2
	PriorityArray  PriArrayStrings `json:"priorityArray"`
	TxnSource      string          `json:"txn_source"`
	TxnNumber      string          `json:"txn_number"`
	ApiTimeout     int             `json:"api_timeout,omitempty"`
	Timeout        int             `json:"timeout,omitempty"`
}
```

### Questions:

- Will the new approach be similar to the current read/write body, with the following differences?  
  - **For read operations:** `objectType`, `objectInstance`, and `property` will be placed inside the `objects` array, while other properties remain unchanged.  
  - **For write operations:** `objectType`, `objectInstance`, `property`, and `value` will be placed inside the `objects` array, with other properties staying the same.  
  - The key requirement is to include `txn_source` and `txn_number` for tracking the results.  

- Would a bulk call be more efficient than making concurrent calls in `driver-bacnet`?  
  - If so, could you provide some details to help us assess whether we should adopt the bulk approach or continue with concurrent calls?

shomaglasang, 2025-04-03:
@RaiBnod do you have a test script to measure the throughput of concurrent calls? I think we can tweak it a bit and re-use to test the bulk calls.

I'll create a request/response body structure like you posted above. I didn't write the last one. Maybe @NubeDev wrote it from the sample MQTT messages I sent.

shomaglasang, 2025-04-06:
@RaiBnod @NubeDev  I will create a simple benchmark program to measure the performance/throughput of the bacnet server when processing between bulk and concurrent calls.

shomaglasang, 2025-09-03:
Hi @RaiBnod have you tried this function already? Pls let me know if there's anything to be added. Thanks.