NubeIO/driver-bacnet #137: BACnet server updates

NubeDev, 2025-04-12:
## multi state point
The bacnet server always has around 10 multistate points which are not used. Please remove these 

## Point fault
Add in if not extsting a way when we write a point value to put it into fault and let use add a fault message

```
{
fault: true
message:"failed to write"
}
```

shomaglasang, 2025-05-02:
@NubeDev Hi mate, do you know what property is the "state" you are showing above for Binary Values (BV)? Is it `event state (36)` or `out of service (81)`

NubeDev, 2025-05-03:
I think its the out-of-service: TRUE

Set that to FALSE

shomaglasang, 2025-05-04:
@NubeDev mate, I create an rc for this [v1.0.3-rc2](https://github.com/NubeIO/driver-bacnet/releases/tag/v1.0.3-rc2).
- Removed multistate points.
- Added "fault" field in error response message (json payload)
- Set default value of "out-of-service" to false.
- I verified the bacnet server can use a different listening port by setting it in the config file.