NubeIO/driver-bacnet #138: Points in config file

NubeDev, 2025-04-12:
We want more control on on the bacnet-server points that are created.

## creating the bacnet point
In the config file

rather then this
```ymal
bi_max: 10
bo_max: 10
bv_max: 20
ai_max: 10
ao_max: 10
av_max: 20
```

Lets update it to 

```ymal
use_points_list: true // if this is true then ignore "bi_max: 10 and so on"
points: 
  AV-1
    name: "my av-1"
    units: "no-units" 
  AV-2
    name: "my av-2"
    units: "no-units" 
```

@shomaglasang if you have a better way to do the ymal thats fine, above is just an example

shomaglasang, 2025-04-30:
@NubeDev @RaiBnod Tested on office device (192.168.15.17). Looks good. 
The points config looks like below:
```
use_points_list: true
points:
  - point: ai-1
    name: my ai-1
    units: no-units
  - point: ai-2
    name: my ai-2
    units: percent
  - point: ao-1
    name: my ao-1
    units: no-units
  - point: ao-2
    name: my ao-2
    units: no-units
  - point: ao-3
    name: my ao-3
    units: no-units
  - point: av-1
    name: my av-1
    units: no-units
  - point: av-2
    name: my av-2
    units: percent
  - point: bi-1
    name: my bi-1
    units: no-units
  - point: bi-2
    name: my bi-2
    units: percent
  - point: bo-1
    name: my bo-1
    units: no-units
  - point: bo-2
    name: my bo-2
    units: no-units
  - point: bo-3
    name: my bo-3
    units: no-units
  - point: bv-1
    name: my bv-1
    units: no-units
  - point: bv-2
    name: my bv-2
    units: percent
```

The format of the point name is validated. The format must be:
```
<object_name>-<instance ID>
Ex: ai-1
```
Valid object_names are: ai, ao, av, bi, bo, bv
The server throws an error if the object name is invalid.
```
- Invalid Object name: [ax]
```
The value of the instance ID is NOT validated. Internally, the object instances have a fixed ID or index, starting from 1, 2, and so on. 

The value of **units** is validated. The server throws an error if the unit is invalid.
```
- Invalid Units: [no-unitsx]
```
You can find the valid units in this file [bacnet units](https://github.com/NubeIO/driver-bacnet/blob/master/bacnet-stack/src/bacnet/bactext.c)

Sample units
```
no-units
parts-per-million
parts-per-billion
percent
```

shomaglasang, 2025-04-30:
Mate @NubeDev let me know when I can merge them into master branch and what's the release version . Thanks.

shomaglasang, 2025-05-18:
Hi mate @NubeDev Please let me know if I can merge this into master branch. And also https://github.com/NubeIO/driver-bacnet/issues/137. Thanks.