NubeIO/driver-bacnet #139: Support IP Address or Interface Name

NubeDev, 2025-06-07:
## Summary
bip_set_interface() currently expects an interface name (like "eth0"), but users may also pass an IP address (like "192.168.15.10"). It would be helpful if the function could:

1. Detect if the input is an IP or interface name
2. Attempt to bind accordingly
3. Return an error only if the IP is not assigned locally

## Why
On systems like Raspberry Pi, interfaces can have multiple IPs (static + DHCP), which causes IP-based binding to fail. Adding detection would improve usability and reliability.

I think this is the code

https://github.com/NubeIO/driver-bacnet/blob/50fdd6b1aac2bc08c19d35d3a9cdf4056d2a1c7a/bacnet-stack/src/bacnet/datalink/bip.h#L50

shomaglasang, 2025-06-09:
Hi mate @NubeDev just want to make sure my understanding is correct. Do you want the driver-bacnet to bind to the IP address if specified, instead of ANY (e.g. INADDR_ANY or listens to ALL)? The IP address will also be used as the source IP address of outgoing UDP bacnet messages. Am I right mate?

NubeDev, 2025-06-09:
Correct i want to bind it to an actual IP

And if the IP dosnt exist it will not work