NubeIO/driver-bacnet #140: Support for YAML-based Dynamic BACnet Schedule Configuration (7-day)

NubeDev, 2025-06-08:
Support for **dynamic creation and configuration of multiple BACnet Schedule objects** using a YAML file.

### Goals

* Support **multiple** `Schedule` objects.
* Use a **YAML file** to define schedules.
* Only create a schedule in the BACnet server if `enable: true`.
* YAML should allow **7-day weekly schedules**, each with multiple time ranges per day.

---

### Example YAML Format

```yaml
schedules:
  - name: "Office AC"
    enable: true
    object_instance: 1
    weekly_schedule:
      Monday:
        - start: "08:00"
          stop: "18:00"
      Tuesday:
        - start: "08:00"
          stop: "18:00"
  - name: "Lobby Lights"
    enable: false
    object_instance: 2
    weekly_schedule:
      Friday:
        - start: "17:00"
          stop: "23:00"
```

---

###  Implementation Outline

1. **Parse the YAML** using a C YAML parser (e.g., libyaml).
2. Iterate over the `schedules` array.
3. For each entry:

   * If `enable: false` → **skip** creation.
   * If `enable: true` → create `Schedule` object using `bacnet-stack` APIs.
4. Populate `Weekly_Schedule` using start/stop time pairs for each day.
5. Register the schedule object using existing stack functionality:

   * Based on `bacnet/basic/object/schedule.c`
   * Register in `device.c` via custom logic or a manager module

## BACnet Runtime Editing (via BMS)
Once created, Schedule objects must support full BACnet read/write of properties.

Support the following properties at minimum:

- Weekly_Schedule (writeable)
- Object_Name
- Present_Value
- Schedule_Default

This allows external BACnet clients (e.g., building automation software) to override or update the schedule at runtime.

YAML-initialized values act as a default or bootstrap only; any BACnet-side updates persist in memory (optionally saved to disk if persistence is added later).

###  Relevant Stack Files

* `bacnet/basic/object/schedule.c`
* `bacnet/basic/object/schedule.h`
* `examples/server/device.c` (for object registration)
* Possibly `bacnet/basic/services.c` (for property handling)

shomaglasang, 2025-08-31:
Hi mate @NubeDev Just working on this. I overlooked this issue.
Got a couple of question.
1. Do we need to support (read/write) via MQTT commands as well?
2. The current implementation supports writing to PROP_OUT_OF_SERVICE property only. So will support writing to the following values:

Weekly_Schedule - PROP_WEEKLY_SCHEDULE (123)
Object_Name - PROP_OBJECT_NAME (77)
Present_Value - PROP_PRESENT_VALUE (85)
Schedule_Default - PROP_SCHEDULE_DEFAULT (174)

NubeDev, 2025-09-01:
let me think about this we may put in on hold for now

closed 2025-09-01 by NubeDev